SpiceSharp/SpiceSharpParser #12: Add PSpice specific evaluator / decoder

ewiebe1, 2018-09-13:
Please add PSpice specific evaluator / decoder

If you add PSpice specific code, it will result in an error, e.g.:

V_V1 N14847 0
+PULSE 0V 5V 3.67us 40ns 40ns 3.255us 6.51us
V_V2 N14651 0 5Vdc
R_R1 N18055 N14538 2.2k TC=0,0
R_R2 N17129 N14555 2.2k TC=0,0
C_C1 N14538 N14569 1n TC=0,0
C_C2 N14555 N14581 330p TC=0,0
R_R3 N14677 N14651 1k TC=0,0
R_R4 N14677 0 1k TC=0,0
R_R5 N14538 N14569 68k TC=0,0
R_R6 N14555 N14581 120k TC=0,0
C_C3 N14569 N17129 10n TC=0,0
X_U6 N14677 N14538 N14651 0 N14569 MCP651
X_U7 N14677 N14555 N14651 0 N14581 MCP651
C_C4 N14847 N18055 100n TC=0,0
V_V3 M_UN0001 0 AC 0
+SIN 2.5V 2.06V 153600 0 0 0
...

[MCP651.txt](https://github.com/SpiceSharp/SpiceSharpParser/files/2380783/MCP651.txt)

marcin-golebiowski, 2018-09-24:
@ewiebe1  I've started to work on that features.

marcin-golebiowski, 2018-10-25:
@ewiebe1 Following features have been implemented:
* TABLE syntax (ABM)
* POLY(n) syntax (ABM)
* VALUE = {expression}
* TC = TC1, TC2 for resistor

I will continue to work on ability to parse and run PSpice netlists. 
Is there any special netlist that you would like to parse?

Marcin

ewiebe1, 2018-11-14:
Hello Marcin, many thanks for your work!

Today, I have made an other test:
I have downloaded the current version of SpiceSharpParser and created a small Test Application. In this test application I go through all *.cir files from the examples folder of ngspice28. 
There were many Exceptions while parsing these files.

My code snippet:
private void TestNetlist(string path_netlist)
{
       try
       {
           SpiceParserSettings sps = new SpiceParserSettings();
           sps.WorkingDirectory = path_netlist.Substring(0, path_netlist.LastIndexOf("\\"));
           SpiceParser sp = new SpiceParser(sps);
           string netlist = File.ReadAllText(path_netlist);
           sp.ParseNetlist(netlist);
       }
       catch (Exception Ex)
       {
           AddLog("Error with " + path_netlist + " : " + Ex.Message);
       }            
}
 
See attached files. 
Thank you for your work again!

[parsing log of ngspice 28.txt](https://github.com/SpiceSharp/SpiceSharpParser/files/2580025/parsing.log.of.ngspice.28.txt)

[ngspice examples.zip](https://github.com/SpiceSharp/SpiceSharpParser/files/2580026/ngspice.examples.zip)

marcin-golebiowski, 2018-11-14:
@ewiebe1 I will work also on ability to parse ngspice28 netlists. Thank you for examples.
I will try my best, but I will take time to write all necessary features to SpiceSharpParser.

Regarding the netlist you posted (MCP651.txt) it's quite close to successful parse. 
In the next release of SpiceSharp/SpiceSharpParser we are adding in particular:
* TC1, TC2 for capactor.
* PWL waveform
So it will should work then.

However I think there is an error in MCP651.txt regarding brackets for TABLE (line 138).

marcin-golebiowski, 2019-02-10:
@ewiebe1 Do you have any other requests regarding SpiceSharpParser?

ewiebe1, 2020-04-08:
I found out today that LTSpice uses also some special features in its SPICE netlists, like "A-Devices".
See here: [Undocumented LTSpice parser features](http://ltwiki.org/index.php?title=Undocumented_LTspice)

Perhaps the SpiceSharpParser should also be able to parse such special LTSpice netlists ?

marcin-golebiowski, 2020-04-08:
@ewiebe1 
I think it would be not difficult to parse that. However much harder is to simulate that components.

svenboulanger, 2020-04-10:
I agree with @marcin-golebiowski . Supporting these custom models is difficult because they'd have to be reverse-engineered. Instead, it may be better to just roll out custom models.

I currently have my hands full with Spice# and Spice#.Behavioral v3, so I don't see this implemented in the near future.