SpiceSharp/SpiceSharpParser #104: Method not found: 'Boolean SpiceSharp.Circuits.Entity.SetParameter(System.String, Double, System.Collections.Generic.IEqualityComparer`1)'

user4githvb, 2019-07-03:
I have issue with diode example (from tutorial):

        var netlist = string.Join(Environment.NewLine,
            "Diode circuit",
            "D1 OUT 0 1N914",
            "V1 OUT 0 0",
            ".model 1N914 D(Is=2.52e-9 Rs=0.568 N=1.752 Cjo=4e-12 M=0.4 tt=20e-9)",
            ".DC V1 -1 1 10e-3",
            ".SAVE i(V1)",
            ".END");

        // Parsing part - SpiceSharpParser
        var parser = new SpiceParser();
        var parseResult = parser.ParseNetlist(netlist);
        var spiceSharpModel = parseResult.SpiceSharpModel;

Method not found: 'Boolean SpiceSharp.Circuits.Entity.SetParameter(System.String, Double, System.Collections.Generic.IEqualityComparer`1)'

user4githvb, 2019-07-03:
Solved - It occurs that SpiceSharp Parser nuget includes a version of SpiceSharp that is older then SpiceSparp standalone nuget that I also installed - even before SpiceSharp Parser.
Newest version of SpiceSparp is in conflict with Parser that required older package.

marcin-golebiowski, 2019-07-05:
@user4githvb  Should be fine now. New version of Spice#.Parser has been published.

user4githvb, 2019-07-05:
Now parser works fine - thank You thank You very much.

closed 2019-07-05 by user4githvb

marcin-golebiowski, 2019-07-05:
@user4githvb If you have any problems, feel free to write e-mail to me or write here.