SpiceSharp/SpiceSharpParser #114: Why can't this circuit be parsed?

DZychlinski, 2019-10-28:
Hi,

I have this circuit (here it's very simplified, but illustrates the problem I face):

```
I_M 0 N1 0.025
X_entry1 N1 0 N2 entry_1
X_entry2 N1 0 N3 entry_2

.subckt entry_1 m_in m_out v_vel
Vmas m_in msx 0  
Rmas msx msy 1e-6
Bxm msy m_out V=V(v_vel)*V(v_vel)/2.1
Hmss mss 0 Vmas 1
Buv 0 v_vel I=(v(mss)/120)/(0.1)
Ruv 0 v_vel 1
.ends

.subckt entry_2 m_in m_out v_vel
Vmas m_in msx 0  
Rmas msx msy 1e-6
Bxm msy m_out V=V(v_vel)*V(v_vel)/2
Hmss mss 0 Vmas 1
Buv 0 v_vel I=(v(mss)/120)/(0.1)
Ruv 0 v_vel 1
.ends

.OP
.control
.options abstol=1e-5
run
.endc
.end

```

SpiceSharpParser is unable to parse division by 2.1 on line
`Bxm msy m_out V=V(v_vel)*V(v_vel)/2.1`
throwing exception:
`SpiceSharpParser.Lexers.LexerException: 'Can't get next token from text: '/2.1
''`

Is there anything that I do wrong here?

marcin-golebiowski, 2019-10-29:
@DZychlinski  I will take a look later today. But I think you should use following syntax:

Bxm msy m_out V={V(v_vel)*V(v_vel)/2.1}

marcin-golebiowski, 2019-10-29:
Also please remove:

.control
run
.endc

DZychlinski, 2019-10-29:
I've introduced your suggestions and now it parses correctly, but it cannot create component of unreported type, throwing this exception:

```
System.Exception: 'Unsupported component type'
   at SpiceSharpParser.ModelReaders.Netlist.Spice.Readers.ComponentReader.GetComponentGenerator(IReadingContext context, String componentName, String& componentType)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.Readers.ComponentReader.Read(Component statement, IReadingContext context)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.Readers.StatementReader`1.Read(Statement statement, IReadingContext context)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.SpiceStatementsReader.Read(Statement statement, IReadingContext readingContext)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.Readers.EntityGenerators.Components.SubCircuitGenerator.CreateSubcircuitComponents(SubCircuit subCircuitDefinition, IReadingContext subCircuitContext)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.Readers.EntityGenerators.Components.SubCircuitGenerator.Generate(String componentIdentifier, String originalName, String type, ParameterCollection parameters, IReadingContext context)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.Readers.ComponentReader.Read(Component statement, IReadingContext context)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.Readers.StatementReader`1.Read(Statement statement, IReadingContext context)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.SpiceStatementsReader.Read(Statement statement, IReadingContext readingContext)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.Context.ReadingContext.Read(Statements statements, ISpiceStatementsOrderer orderer)
   at SpiceSharpParser.ModelReaders.Netlist.Spice.SpiceNetlistReader.Read(SpiceNetlist netlist)
   at SpiceSharpParser.SpiceParser.ParseNetlist(String spiceNetlist)
```

marcin-golebiowski, 2019-10-29:
Strange. Can I help you with that in the evening?

marcin-golebiowski, 2019-10-29:
Are you using NuGet package or using source code directly ?

DZychlinski, 2019-10-29:
NuGet package as of now, version 1.2.0.

marcin-golebiowski, 2019-10-29:
Please do the following for now if you don't have time till evening:
Download source code of SpiceSharpParser from master
First try to reference SpiceSharpParser from source instead of nuget package.

Please let me know if that helped.

DZychlinski, 2019-10-29:
I've downloaded SpiceSharpParser source, referenced it in my project and it actually did parse my netlist succesfully. :)
Then I wanted to run my full circuit but I had different problem this time: `"Unrecognized Spice property 'SpiceSharpBehavioral.Parsers.SpiceProperty' at character 7 ('/')"`.
I downloaded sources for SpiceSharp, referenced it and got more info on my error by debugging it. Turned out I had some function calls written in lower-case and after fixing those - simulation finished without problems.

I know not everything here is related to SpiceSharpParser, but I thought it would be nice to provide some feedback in case someone else would stumble upon similar issue.

Thank you very much for your support!

marcin-golebiowski, 2019-10-29:
@DZychlinski If you need any feature, please tell me. I'm open for any feature suggestion in SpiceSharpParser.

DZychlinski, 2019-10-29:
Well, if I could suggest something, that would be my list:

1. exception messages could have messages providing more information - of course if it's viable in given context, like entire expression that is causing problems (not just part of it, as that might occur in many more places);
2. it would be really nice if Parser could take care of equations that now require curly brackets, so that it would parse them without the need to add them;
3. I don't quite know why Parser requires those to be removed, but maybe it would be possible for it to build circuit despite having those statements?:
> .control
> run
> .endc
4. I cannot reproduce it now, but if I remember correctly, when Parser informs about an "error on line X" it does so without counting comments lines, so we get line that doesn't match source NetList. This is connected to point 1, so just a better message would suffice (at least for me).

I don't know how feasible those suggestions are, but they are quite good quality of life improvements and they would allow to parse pure NetLists without any changes.

marcin-golebiowski, 2019-10-29:
@DZychlinski Thank you for feedback. I will work on those suggestions.

DZychlinski, 2019-10-30:
@marcin-golebiowski Oh, one more thing that might be useful for achieving more compatibility with SPICE NetLists is that you can define multiline statements with '+' sign at the very beginning of the next, continued line, like so:
```
Bxm msy m_out V=V(v_vel)
+ *V(v_vel)/2.1
```

marcin-golebiowski, 2019-11-01:
@DZychlinski Nuget package v1.3.0 for  SpiceSharpParser has been published.

It's able to parse and simulate following your netlist with small modifications:

`Real life example

I_M 0 N1 0.025
X_entry1 N1 0 N2 entry_1
X_entry2 N1 0 N3 entry_2

.subckt entry_1 m_in m_out v_vel
Vmas m_in msx 0  
Rmas msx msy 1e-6
Bxm msy m_out V={V(v_vel)*V(v_vel)/2.1}
Hmss mss 0 Vmas 1
Buv 0 v_vel I={(V(mss)/120)/(0.1)}
Ruv 0 v_vel 1
.ends

.subckt entry_2 m_in m_out v_vel
Vmas m_in msx 0  
Rmas msx msy 1e-6
Bxm msy m_out V={V(v_vel)*V(v_vel)/2}
Hmss mss 0 Vmas 1
Buv 0 v_vel I={(V(mss)/120)/(0.1)}
Ruv 0 v_vel 1
.ends

.OP
.SAVE V(N1)
.end`

I will work on improvements.
Please let me know if you have any other needs.

DZychlinski, 2019-11-05:
Thank you very much for you efforts!
As for me this issue can be closed. I have one more thing I' having problems with, but that belongs to SpiceSharpBehavioral - I'll ask it there.

closed 2019-11-06 by marcin-golebiowski