SpiceSharp/SpiceSharpParser #139: Component Generators that throw ReadingException should pass netlist text file line number

ToddThomson, 2019-12-19:
The Resistor generator function may throw:
```
throw new ReadingException("Invalid parameter for resistor: " + parameter.Image);
```
This should include the ```parameters``` (netlist.Text) line # for better feedback to the end user or API call.

In general, the netlist source text (file) line number should be included, where possible.

It would be appropriate for the lexer/parser to keep column properties as well. Netlist editors would then be able to highlight issues with a better user experience then provided.

marcin-golebiowski, 2019-12-19:
@ToddThomson 
This is some leftover that I've missed in recent change to include line number information .  That will be fixed today.

Regading column number. I think that is a good idea and that will be done in next few days (maybe hopefully also today).

ToddThomson, 2019-12-19:
@marcin-golebiowski Great. Thanks!

marcin-golebiowski, 2019-12-19:
#143

marcin-golebiowski, 2019-12-19:
New NuGet package 1.4.4. has been published with things that you asked.
However there can be some bugs in this area. Please report them and I will fix them.

ToddThomson, 2019-12-19:
@marcin-golebiowski You're a nimble programmer! Nice. I will pull the changes and test things out.

marcin-golebiowski, 2021-04-28:
If you have any other issues with line numbers, please reopen the issue.

closed 2021-04-28 by marcin-golebiowski