SpiceSharp/SpiceSharpParser #141: Netlist Inline comment

ToddThomson, 2019-12-19:
It appears that "*" and ";" comments are supported. Comments can either be whole line or at some column in the statement line.

When a statement is properly formed and ends in a comment then the netlist source text validates.

However, with improperly formed statements with an inline comment things go badly.

I recommend that everything after the '*' or ";" be ignored by the lexer. I would also keep a ```Comment``` starting and ending location for each statement, if applicable ( you may already have this ).

marcin-golebiowski, 2019-12-19:
I will take a look at this

ToddThomson, 2019-12-19:
@marcin-golebiowski '*' is recognized as a whole line comment. It must occur as the first char on a given line. If the '*' is not the first char then it will be interpreted as a WORD.

The ';' char works as the start of a comment and can be used anywhere.

If this is intended then fine. However, it should be documented or the '*' should be used as a comment block start too.

ToddThomson, 2019-12-20:
If ```*``` can be used within a math expression then just use  ```;``` for comments at the end of a statement.

marcin-golebiowski, 2021-04-28:
@ToddThomson 

Fixed in commit [c418859531f81e65ab4418b2cd6ee275a01145a1](https://github.com/SpiceSharp/SpiceSharpParser/commit/c418859531f81e65ab4418b2cd6ee275a01145a1)

marcin-golebiowski, 2021-04-28:
If you have any other request to comments, please reopen the issue.

closed 2021-04-28 by marcin-golebiowski