SpiceSharp/SpiceSharpParser #142: API Return Results Vs Exceptions

ToddThomson, 2019-12-19:
It would be nice to have the Parser/Validator API return an object with full file parsing/validation errors ( or competed successfully ) rather than halting at the first exception.

I believe you do this with "Warnings" in the circuit model. (NOTE: do simulations run with warnings? ).

marcin-golebiowski, 2019-12-19:
@ToddThomson 
It will be done in few days. I will let you know when it's ready.

ToddThomson, 2019-12-19:
@marcin-golebiowski Thanks Marcin.

```
    public class SpiceServiceResult
    {
        #region Fields

        private static readonly SpiceServiceResult _success = new SpiceServiceResult( true );

        #endregion

        #region Constructors

        private SpiceServiceResult( bool success )
        {
            Succeeded = success;
        }

        public SpiceServiceResult( SpiceServiceError error )
        {
            Succeeded = false;
            Error = error;
        }

        /// <summary>
        /// Creates a new <see cref="SpiceServiceResult"/> with the provided <see cref="EntityServiceError"/> error.
        /// </summary>
        /// <param name="errorType"></param>
        /// <param name="error"></param>
        public SpiceServiceResult( SpiceServiceErrorType errorType, SpiceServiceError error )
        {
            Succeeded = false;
            ErrorType = errorType;
            Errors = new SpiceServiceErrorCollection();

            Errors.AddError( error );
        }

        /// <summary>
        /// Creates a new <see cref="SpiceServiceResult"/> with the provided <see cref="SpiceServiceErrorCollection"/> errors.
        /// </summary>
        /// <param name="errorType"></param>
        /// <param name="errors"></param>
        public SpiceServiceResult( SpiceServiceErrorType errorType, SpiceServiceErrorCollection errors )
        {
            Succeeded = false;
            ErrorType = errorType;
            Errors = errors;
        }

        #endregion

        #region Methods

        /// <summary>
        /// Creates a failed <see cref="SpiceServiceResult"/> with the <see cref="Exception"/> cause.
        /// </summary>
        /// <param name="errorType"></param>
        /// <param name="e"></param>
        /// <returns></returns>
        public static SpiceServiceResult Failed( SpiceServiceErrorType errorType, Exception e = null )
        {
            return new SpiceServiceResult( errorType, new SpiceServiceError( string.Empty, e ) );
        }

        /// <summary>
        /// Creates a failed <see cref="SpiceServiceResult"/> with the <see cref="EntityServiceError"/> cause.
        /// </summary>
        /// <param name="errorType"></param>
        /// <param name="error"></param>
[... 206 lines cut ...]
        IEnumerator IEnumerable.GetEnumerator()
        {
            return Errors.GetEnumerator();
        }
    }
```
```
    public enum SpiceServiceErrorType : int
    {
        /// <summary>
        /// Indicates an unknown error type.
        /// </summary>
        Unknown = 0x01,

        /// <summary>
        /// Indicates entity validation error type.
        /// </summary>
        Validation = 0x02,

        /// <summary>
        /// Indicates a repository error type.
        /// </summary>
        Parsing = 0x03,
    }
```

marcin-golebiowski, 2019-12-19:
@ToddThomson 
If you prefer/want feel free to open PR or I will use your code in the repository.

ToddThomson, 2019-12-19:
@marcin-golebiowski Go ahead and use what you can to speed things along. I am busy working through the simulator code now.

marcin-golebiowski, 2019-12-20:
@ToddThomson Just to let you know about ETA for this: 31 December

ToddThomson, 2019-12-20:
@marcin-golebiowski Hey, we all need holiday time! I look forward to seeing the changes.

marcin-golebiowski, 2019-12-20:
@ToddThomson I will try to implement changes today but I'm not sure if I make it.

marcin-golebiowski, 2019-12-20:
Please review #147.
It's what I was able to do in 2h.

ToddThomson, 2019-12-20:
@marcin-golebiowski Deployment to NuGet is not really necessary. I use VS2019 GitHub services to pull your changes into my local repo. It is also likely a good idea to limit the changes to the released Nuget package version. Perhaps a dev branch would be in order?
 When working on a project (always in debug mode) that is in development stage I use the following within csproj: 
```
<Choose>
    <When Condition="'$(Configuration)' == 'Debug'">
      <ItemGroup>
        <ProjectReference Include="..\..\SpiceSharp\SpiceSharp\SpiceSharp.csproj" />
      </ItemGroup>
    </When>
    <Otherwise>
      <ItemGroup>
        <PackageReference Include="SpiceSharp" Version="2.8.0" />
      </ItemGroup>
    </Otherwise>
  </Choose>
```

ToddThomson, 2019-12-20:
If you create a dev branch for all your SpiceSharp repos I will hook up my VS2019 to use them ( rather than master ).

marcin-golebiowski, 2019-12-20:
Development branch has been created. 

I thought that Nuget packages every few days were good idea. They were backward compatible so new features where added in new versions.

ToddThomson, 2019-12-20:
Yeah, no issues with creating NuGet packages that don't change the API. Better to push to Master when you know you are not going to break customer apps that depend on your library!

marcin-golebiowski, 2019-12-21:
OK. I will use development branch from today

marcin-golebiowski, 2020-01-19:
@ToddThomson 1.5.0 version available on Nuget contains requested changes

ToddThomson, 2020-01-19:
@marcin-golebiowski , @svenboulanger Thank-you. I will incorporate your changes this coming week.

I have made progress on the Spice# Studio app. UWP is an interesting challenge given its WinRT beginnings. However, MS is correcting its mistakes and with Win UI  3 coming this year, UWP will be equipped to build desktop apps.

My main challenge was to build a page navigation service to support multiple windows. The Spice# Studio app supports 3 main panes: LHS Main/Solution tool panes based on the standard UWP Navigation View; Central Tabbed documents pane; RHS tool pane based on a custom control which utilizes the Navigation view. There is also a main top menu above a command toolbar and a lower status bar.

I should be able to get a beta pushed up to Github in the next 2 to 3 weeks.

ToddThomson, 2020-01-20:
@svenboulanger I would like to be able to run the parser on an async background thread as the user is actively editing the netlist. Do you have any plans for providing an async API?

marcin-golebiowski, 2020-01-20:
@ToddThomson On SpiceSharpParser level I could provide some async API for simulation that uses events to deliver async result.

ToddThomson, 2020-01-20:
@marcin-golebiowski @svenboulanger The SpiceParserValidationResult looks good! Thank-you.

The Spice# Studio app has a service for interaction with the Spice# library modules. It has a BuildAsync( netlist, warningsAsErrors ) method. Internally the builder calls ParseNetList() and then if no errors are encountered, calls model.Circuit.Validate();

The circuit Validate() call doesn't return anything other that an exception if there is a problem. Was this going to change? A SpiceModelValidationResult or SpiceCircuitValidationResult would seem appropriate. I can't recall if this was being worked on or not. Please let me know.

ToddThomson, 2020-01-20:
@svenboulanger To determine if a validation error or warning occurred during the ParseNetlist() call, one needs to check each Validation Source...
```
        public SpiceParserValidationResult()
        {
            Lexing = new LexerValidationResult();
            Parsing = new ParsingValidationResult();
            Reading = new SpiceNetlistValidationResult();
        }
```
Given that a ValidationEntry has a Source property, it is not necessary to have individual collections. It might be better to have a single collection and then properties to return All and filtered results for the individual Sources.

svenboulanger, 2020-01-20:
Hi @ToddThomson ,

> I would like to be able to run the parser on an async background thread as the user is actively editing the netlist. Do you have any plans for providing an async API?

I am not planning on adding an async API for Spice# (core). You can always run Spice# simulations in a background thread if you don't wan't the UI to freeze. Spice# v3 will have support for multithreaded computations, but that is really only meant for performance optimizations, and is usually very dependent on the circuit.

> The circuit Validate() call doesn't return anything other that an exception if there is a problem. Was this going to change?

Currently, Spice# v3 validation will be done by running a set of rules on subjects. After validation completes, all rule violations can be enumerated (and therefore be filtered, logged, etc.). Additionally, a simulation runs its own set of strictly necessary rules (if rule checking for that simulation is not disabled), and will throw an exception if there are rule violations. I did not really want to distinguish between errors and warnings for rules because that would be context-dependent.

Summarized I would describe the validation system as:
`IRules` contains multiple `IRule` definitions. You can feed these rules `IRuleSubject` instances (usually circuit components) that each `IRule` *can* use to find rule violations. After you have fed all subjects to the rules, you can enumerate all `IRuleViolation`s that each specify the rule that was violated, and optionally the `IRuleSubject` that caused the rule violation.

ToddThomson, 2020-01-20:
@svenboulanger Yeah, don't worry about the async API. I will add the async calls as needed into the SpiceSharp service layer.

I look forward to seeing the new features in Spice# v3!

marcin-golebiowski, 2020-01-29:
@ToddThomson I removed derived validation collection classes #150. It's not yet merged into master. 
Please let me know what think.

ToddThomson, 2020-01-30:
@marcin-golebiowski I'll be able to take a look next week at some point. Thanks!

marcin-golebiowski, 2021-05-27:
Changed by #152.

The validation result collection was simplified.

closed 2021-05-27 by marcin-golebiowski