SpiceSharp/SpiceSharpParser #157: Package install error

bharatsangale1, 2022-05-09:


bharatsangale1, 2022-05-10:
Same error with 4.6.1 and 4.8

Could not install package 'SpiceSharp-Parser 3.1.3'. You are trying to install this package into a project that targets '.NETFramework,Version=v4.6.1', but the package does not contain any assembly references or content files that are compatible with that framework. For more information, contact the package author.

Could not install package 'SpiceSharp-Parser 3.1.3'. You are trying to install this package into a project that targets '.NETFramework,Version=v4.8', but the package does not contain any assembly references or content files that are compatible with that framework. For more information, contact the package author.

marcin-golebiowski, 2022-05-10:
This is strange.  I've created simple console app (.NETFramework 4.8) with SpiceSharp-Parser 3.1.3 and it works. 
This is my packages.config file:

```
<?xml version="1.0" encoding="utf-8"?>
  <packages>
    <package id="Microsoft.CodeAnalysis.Analyzers" version="3.3.2" targetFramework="net48" developmentDependency="true" />
    <package id="Microsoft.CodeAnalysis.Common" version="3.11.0" targetFramework="net48" />
    <package id="Microsoft.CodeAnalysis.CSharp" version="3.11.0" targetFramework="net48" />
    <package id="SpiceSharp" version="3.1.5" targetFramework="net48" />
    <package id="SpiceSharpBehavioral" version="3.1.3" targetFramework="net48" />
    <package id="SpiceSharp-Parser" version="3.1.3" targetFramework="net48" />
    <package id="System.Buffers" version="4.5.1" targetFramework="net48" />
    <package id="System.Collections.Immutable" version="5.0.0" targetFramework="net48" />
    <package id="System.Memory" version="4.5.4" targetFramework="net48" />
    <package id="System.Numerics.Vectors" version="4.5.0" targetFramework="net48" />
    <package id="System.Reflection.Emit.Lightweight" version="4.7.0" targetFramework="net48" />
    <package id="System.Reflection.Metadata" version="5.0.0" targetFramework="net48" />
    <package id="System.Runtime.CompilerServices.Unsafe" version="5.0.0" targetFramework="net48" />
    <package id="System.Text.Encoding.CodePages" version="4.5.1" targetFramework="net48" />
    <package id="System.Threading.Tasks.Extensions" version="4.5.4" targetFramework="net48" />
  </packages>
```

I need more time to help you. I will take a look at it again later today.

bharatsangale1, 2022-05-10:
I used VS2015 but tried on VS2019. It installed successfully on VS2019.

closed 2022-05-10 by bharatsangale1