SpiceSharp/SpiceSharpParser #159: Parser failure when try to extract the voltage at node

betty2310, 2024-04-22:
I try to export voltage at node `out` in the netlist provided on README
```csharp
var netlistText = string.Join(Environment.NewLine,
                "Diode circuit",
                "D1 OUT 0 1N914",
                "V1 OUT 0 0",
                ".model 1N914 D(Is=2.52e-9 Rs=0.568 N=1.752 Cjo=4e-12 M=0.4 tt=20e-9)",
                ".DC V1 -1 1 10e-3",
                ".SAVE v(OUT)",
                ".END");
```

But I get this error `Unhandled exception. System.ArgumentNullException: Value cannot be null. (Parameter 'key')`

Here is full exception message
```
Unhandled exception. System.ArgumentNullException: Value cannot be null. (Parameter 'key')
   at System.Collections.Generic.Dictionary`2.FindValue(TKey key)
   at System.Collections.Generic.Dictionary`2.TryGetValue(TKey key, TValue& value)
   at SpiceSharp.Simulations.VariableDictionary`1.TryGetValue(String key, V& value)
   at SpiceSharp.Simulations.RealVoltageExport.GetVariable(IReadOnlyList`1 path)
   at SpiceSharp.Simulations.RealVoltageExport.Initialize(Object sender, EventArgs e)
   at SpiceSharp.Simulations.Simulation.OnAfterSetup(EventArgs args)
   at SpiceSharp.Simulations.Simulation.Run(IEntityCollection entities)
```

My enviroinment:
+ SpiceSharpParser version 3.1.5
+ Dotnet core version 8.0.200

betty2310, 2024-05-11:
@marcin-golebiowski Hi, sorry for mention without permission but can you please help?

marcin-golebiowski, 2024-05-11:
Hi,
Following snippet works for me:

```
var netlistText = string.Join(
    Environment.NewLine,
    "Diode circuit",
    "D1 OUT 0 1N914",
    "V1 OUT 0 0",
    ".model 1N914 D(Is=2.52e-9 Rs=0.568 N=1.752 Cjo=4e-12 M=0.4 tt=20e-9)",
    ".DC V1 -1 1 10e-3",
    ".SAVE i(V1)",
    ".END");

// Parsing part
var parser = new SpiceNetlistParser();
var parseResult = parser.ParseNetlist(netlistText);
var netlist = parseResult.FinalModel;

// Translating netlist model to SpiceSharp
var reader = new SpiceSharpReader();
var spiceSharpModel = reader.Read(netlist);

// Simulation using SpiceSharp
var simulation = spiceSharpModel.Simulations.Single();
var export = spiceSharpModel.Exports.Find(e => e.Name == "i(V1)");
simulation.ExportSimulationData += (sender, args) => Console.WriteLine(export.Extract());
simulation.Run(spiceSharpModel.Circuit);
```

I use SpiceSpar (3.1.8) and latest SpiceSharpParser

betty2310, 2024-05-12:
@marcin-golebiowski 
Hi, thanks for your reply.
As I mentioned, I get problem when export node is `v(OUT)`. If export `i(V1)`, it's work fine.

> I try to export voltage at node `out` in the netlist provided on README

marcin-golebiowski, 2024-05-12:
I've found the issue. Below is working example.

var netlistText = string.Join(
    Environment.NewLine,
    "Diode circuit",
    "D1 OUT 0 1N914",
    "V1 OUT 0 0",
    ".model 1N914 D(Is=2.52e-9 Rs=0.568 N=1.752 Cjo=4e-12 M=0.4 tt=20e-9)",
    ".DC V1 -1 1 10e-3",
    ".SAVE i(V1) v(OUT,0)",
    ".END");

// Parsing part
var parser = new SpiceNetlistParser();
var parseResult = parser.ParseNetlist(netlistText);
var netlist = parseResult.FinalModel;

// Translating netlist model to SpiceSharp
var reader = new SpiceSharpReader();
var spiceSharpModel = reader.Read(netlist);

// Simulation using SpiceSharp
var simulation = spiceSharpModel.Simulations.Single();
var export = spiceSharpModel.Exports.Find(e => e.Name == "v(OUT,0)");
simulation.ExportSimulationData += (sender, args) => Console.WriteLine(export.Extract());
simulation.Run(spiceSharpModel.Circuit);

closed 2024-05-12 by betty2310